HASwitchPlate/openHASP #899: Makerfabs MaTouch TFT Parallel 5" Support - Initial ini included!

jonofmac, 2025-04-29:
Hello,
I took a stab at creating the .ini file for the MakerFabs S3 TFT Parallel 5" touch screen device.

Product Link: https://www.makerfabs.com/matouch-esp32-s3-parallel-tft-with-touch-5.html

I dug through the schematic and example code and built a .ini file that appears to function for the most part. I am getting some minor screen glitches that I need to figure out, but maybe someone more experienced can help me with it. Touch screen works. Screen works 99% (randomly glitches every few seconds). Wifi works. 

makerfabs-s3-tft50-parallel.ini
```
;***************************************************;
;   Makerfabs ESP32-S3 Parallel TFT With Touch 5"   ;
;               - 5" RGB Parallel TFT               ;
;               - GT911 touch controller            ;
;***************************************************;

[env:makerfabs-s3-tft50-parallel]
extends = arduino_esp32s3_v2, flash_16mb
board = esp32-s3-devkitc-1
board_build.arduino.memory_type = qio_opi

build_flags =
    -D HASP_MODEL="Makerfabs ESP32-S3 Parallel TFT With Touch 5.0"
    ${arduino_esp32s3_v2.build_flags}
    ${esp32s3.ps_ram}

;region -- ArduinoGFX build options ------------------------
    -D HASP_USE_ARDUINOGFX=1
    -D HASP_LV_USE_SW_ROTATE=1
    -D TFT_WIDTH=800
    -D TFT_HEIGHT=480
    ; Bus Settings
    -D TFT_DE=40
    -D TFT_VSYNC=41
    -D TFT_HSYNC=39
    -D TFT_PCLK=42
    -D TFT_R0=45
    -D TFT_R1=48
    -D TFT_R2=47
    -D TFT_R3=21
    -D TFT_R4=14
    -D TFT_G0=5
    -D TFT_G1=6
    -D TFT_G2=7
    -D TFT_G3=15
    -D TFT_G4=16
    -D TFT_G5=4
    -D TFT_B0=8
    -D TFT_B1=3
    -D TFT_B2=46
    -D TFT_B3=9
    -D TFT_B4=1
    -D TFT_BCKL=2
    ; Panel Settings
    -D TFT_HSYNC_POLARITY=0
    -D TFT_HSYNC_FRONT_PORCH=8
    -D TFT_HSYNC_PULSE_WIDTH=4
    -D TFT_HSYNC_BACK_PORCH=8
    -D TFT_VSYNC_POLARITY=0
    -D TFT_VSYNC_FRONT_PORCH=8
    -D TFT_VSYNC_PULSE_WIDTH=4
    -D TFT_VSYNC_BACK_PORCH=8
    -D TFT_PCLK_ACTIVE_NEG=1
    -D TFT_PREFER_SPEED=16000000
    -D TFT_AUTO_FLUSH=1
    ; Touch Settings
    -D TOUCH_WIDTH=800
    -D TOUCH_HEIGHT=480
    -D TOUCH_DRIVER=0x911
    -D TOUCH_SCL=18
    -D TOUCH_SDA=17
    -D TOUCH_IRQ=-1
    -D TOUCH_RST=38
    -D I2C_TOUCH_ADDRESS=0x5D
    -D I2C_TOUCH_FREQUENCY=400000
;endregion

lib_deps =
    ${arduino_esp32s3_v2.lib_deps}
    ${arduinogfx.lib_deps}
    Arduino_RPi_DPI_RGBPanel_mod
    ${goodix.lib_deps}

```

jonofmac, 2025-04-29:
An update: the components seem to work. Glitches seem to only happen while I'm actively touching the screen. It kind of looks like the screen loses sync between the different color channels. Attaching good and bad frame. I tried dropping the screen frequency and that just completely broke the display. 

[image]
[image]

marsman7, 2025-05-01:
How can I recreate this? Can you upload your pages.json here?

jonofmac, 2025-05-01:
You can get the screen to glitch by touching or interacting with any elements. For me, just typing in the wifi password caused it to glitch every 4-5 button presses.

The screen I have here is just putting a lot of random elements up so that I could interact with them. I find grabbing the color selector and sliding it in a circle makes it glitch the most.

Here's my pages file:
```
{"comment":"Designer Config","plate":{"w":320,"h":240,"size_name":"small_landscape","primary_color":"#42a5f6ff","secondary_color":"#fb8c00ff","theme":"materialdark"}}
{"page":0,"comment":"Page 0"}
{"id":1,"obj":"slider","x":40,"y":300,"w":740,"h":20,"pad_top20":0,"pad_right20":0,"pad_left20":0,"pad_bottom20":0,"radius20":14}
{"id":2,"obj":"btn","x":360,"y":400,"w":100,"h":40}
{"id":3,"obj":"spinner","x":140,"y":360,"w":100,"h":100,"angle":45,"line_width":21,"pad_top":8,"pad_right":8,"pad_left":8,"pad_bottom":8,"speed":3800,"type":2,"line_width10":15}
{"id":4,"obj":"cpicker","x":620,"y":340,"w":100,"h":100}
{"id":5,"obj":"btnmatrix","x":60,"y":20,"w":200,"h":200,"options":["1","2","3","\n","4","5","6","\n","7","8","9","\n","*","0","."]}
{"id":7,"obj":"qrcode","x":520,"y":80,"text":"https://openhasp.com","size":100}
{"id":8,"obj":"dropdown","x":320,"y":20,"w":160,"h":50,"options":"Option1\nOption2\nOption3","max_height":140}

```

jonofmac, 2025-05-01:
I made a pages file that does a good job of glitching the screen without any user input:
```
{"page":0,"comment":"Page 0"}
{"id":1,"obj":"spinner","x":20,"y":240,"w":236,"h":236,"angle":45,"line_color10":"#ffffff","pad_top":8,"pad_right":8,"pad_left":8,"pad_bottom":8,"type":2,"line_width":15,"line_width10":15}
{"id":2,"obj":"spinner","x":80,"y":300,"w":121,"h":121,"angle":45,"line_color10":"#ffffff","pad_top":8,"pad_right":8,"pad_left":8,"pad_bottom":8,"speed":600,"type":2,"line_width":15,"line_width10":15}
{"id":3,"obj":"spinner","x":20,"y":0,"w":240,"h":243,"angle":45,"line_width":50,"line_width10":50,"line_rounded10":false,"line_color10":"#ff0000","pad_top":8,"pad_right":8,"pad_left":8,"pad_bottom":8,"type":2}
{"id":4,"obj":"spinner","x":280,"y":0,"w":238,"h":238,"direction":true,"angle":45,"line_width":50,"line_width10":50,"line_rounded10":false,"line_color10":"#00ff00","pad_top":8,"pad_right":8,"pad_left":8,"pad_bottom":8,"speed":1300,"type":2}
{"id":5,"obj":"spinner","x":540,"y":0,"w":240,"h":236,"direction":true,"angle":100,"line_width":70,"line_width10":70,"line_rounded10":false,"line_color10":"#0000ff","pad_top":8,"pad_right":8,"pad_left":8,"pad_bottom":8,"speed":600,"type":2}
{"id":6,"obj":"cpicker","x":580,"y":260,"w":200,"h":200}
{"id":7,"obj":"spinner","x":300,"y":240,"w":240,"h":232,"angle":200,"line_width":50,"line_width10":50,"line_rounded10":false,"pad_top":8,"pad_right":8,"pad_left":8,"pad_bottom":8,"speed":1300,"type":2}
```

https://github.com/user-attachments/assets/bf5b22bb-4727-4d8e-ae08-7c96c34998d2

jonofmac, 2025-05-01:
Interesting, are you using a custom .ini file? Can you point me to the .ini file you're using? Maybe i have an issue with my config somewhere else rather than panel settings.

marsman7, 2025-05-01:
i use the "elecrow-s3-8048c050_4MB" from crowpanel-hmi.ini. But other TFT use other settings.

fvanroie, 2025-05-01:
Try lowering `TFT_PREFER_SPEED` to 8, 10, 12, 14 Mjz... some experimentation could be required.
I think the frequency is too high for pushing that much pixel changes rapidly.
The ESP32 isn't great at fast animations at high speeds, and the board design also matters.
It could be Electromagnetic interference (EMI)...
To me it looks like the pixel clock gets out of sync or can't keep up, resulting in missed data and glitches.

jonofmac, 2025-05-01:
> Try lowering `TFT_PREFER_SPEED` to 8, 10, 12, 14 Mjz... some experimentation could be required. I think the frequency is too high for pushing that much pixel changes rapidly. The ESP32 isn't great [...]

I've tried several different preferred speeds, but anything except 16 MHz breaks the display and I just get red, green, blue, black rotating. No idea why. I've tried 14, 12, 10, 8, 6, 4, 2 MHz, and it's the same behavior for ALL of it. Almost like the S32 device is just creating some default output when the clock doesn't line up with something?

fvanroie, 2025-05-01:
From other configs/displays I typically use these values from the datasheet:
-  **Maximum** HSYNC/VSYNC  Front Porch
- **Typical** HSYNC/VSYNC  Pulse Width
- **Typical** HSYNC Back Porch
- **Half** of Typical DCLK Frequency

So try:
```ini
    -D TFT_HSYNC_POLARITY=0
    -D TFT_HSYNC_FRONT_PORCH=255  // can lower this later to optimize
    -D TFT_HSYNC_PULSE_WIDTH=48
    -D TFT_HSYNC_BACK_PORCH=88
    -D TFT_VSYNC_POLARITY=0
    -D TFT_VSYNC_FRONT_PORCH=255  // can lower this later to optimize
    -D TFT_VSYNC_PULSE_WIDTH=3
    -D TFT_VSYNC_BACK_PORCH=32
    -D TFT_PCLK_ACTIVE_NEG=1
    -D TFT_PREFER_SPEED=15000000  // or 16Mhz if that works better
```

This is all experimental and each device is different.
Extending the Front Porch values might keep the ESP and display in sync better.

jonofmac, 2025-05-01:
The settings resulted in a solid white display. I changed speed back to 16M, still got white. It's worth noting that every time I flash, I unplug and replug the USB to do a full power cycle on the display, because I noticed sometimes settings don't seem to affect the display output until I do that.

1. So I rolled the front porchs back to what was "sort of" working and increased the front porches to the 255 limit: White display. 
2. Tried 200: White display
3. Tried 175: White display
4. Tried 150: Image, but glitching about as much as before
5. Tried 128: Image, but it glitches about as much as before (using same settings as I posted above for other settings)
6. Tried 64: Same as above

So now I figured I should try increasing the back porch values. So I'll leave front at 64 since that leaves some margin with where I was getting white screens. So now varying back porch on horizontal
1. 8: What I've been using. It works but glitches
2. 88 (per datasheet)): No noticeable change from 8.

Now I change vsync back porch:
1. 8: What I've been using. Works but glitches
2. 32 (per datasheet): Same as above. No noticeable difference

Now I try changing the pulse width of hsync:
1. 4: Works but glitches.
2. 48 (Per datasheet): White display
3. 40: White display
4. 32: White display
5. 24: White display
6. 16: White display
7. 8: Functional image, but glitching remains as constant as original settings
8. 1: Functional image, but glitching remains as constant as original settings
9. 1, but I flipped the HSYNC polarity to see if it changes anything: Exactly the same behavior as above 8.

Now I try changing the pulse width of vsync:
1. 3 (per datasheet): Works but glitches
2. 255: Solid color screen rotating through white, black, red, green, and blue
3. 128: Same as above
4. 64: Image works. **I am seeing glitching, but the good news is I'm not seeing the image bouncing up and down during the glitch, it's just color glitching or going solid black**
5. 64, but with the vsync polarity = 1 instead of 0: Same as above. So roll polarity back to 0 for next.
6. 32: The vertical bouncing/glitching is back.
7. 16: Same as above. 
8. 48: Same as above
9. 56: Less vertical bouncing, but not none.

I don't believe the HSYNC or VSYNC polarity does anything. I'm not very familiar with this interface, but what I can find googling seems to show that you either use the HSYNC and VSYNC pins or you use the DE pin to control data latching, so since DE is defined, I guess that makes sense why HSYNC and VSYNC seem to do nothing. I figured I'd try using the SYNC pins instead of DE since the schematic shows it's pinned out, but that did not compile (complained about no DE definition).

**TLDR: Not any real meaningful changes after turning every knob. It almost seems like 16 MHz is just too fast for the display. But I don't understand why anytime I slow the clock down, i get no image.**

I tried changing the clock to 12 MHz as my only change: white screen.

I don't get it....

fvanroie, 2025-05-01:
I don't think it has to do with openHASP firmware, other sketches with the same parameters will give the same results.

fvanroie, 2025-05-01:
Try the examples from their Github, if those don't work... it could be a hardware issue.

jonofmac, 2025-05-01:
Ugh. The demo code works. The hard part is that there's just not enough refreshes to really break it. 

I wrote some basic code to bounce 2 squares around the screen as quick as it can (but still quite slow because their tasks have really slow ticks).

This certainly seems to be something specific to openhasp.

I do see some screen tearing on this example code, but none of the weird color glitching issues, though.

TommySharpNZ, 2025-05-22:
I am running the Elecrow 5" and also see tearing...
Version | 0.7.0-rc12 cd34d89
elecrow-s3-8048c050_4MB

@marsman7 did you do anything special to get yours working relaibly? I am not compiling anything just using the web installed to load one of the pre-compiled binaries.