HASwitchPlate/openHASP #907: Support of  Elecrow Advance 7

Jeremiah-oAT, 2025-05-16:
Any plans to add support for the Elecrow Advance 7? Looking for full openHasp features.

fvanroie, 2025-05-16:
This device is not supported yet and I do not own it.
You can create and submit a configuration for this device via PR.
There are so many new devices coming out that it is not possible to buy, add and test them all myself.

ahb-bs, 2025-05-21:
The new dual-ESP32 offer some incredible screen types/sizes, but I can't work out how to update the code to work across the two (one doing Wifi/admin, the other the screen and touch).

Hopefully if we can crack one then the others should be easier.

Hutchyboi, 2025-05-27:
https://github.com/[Elecrow-RD/CrowPanel-Advance-7-HMI-ESP32-S3-AI-Powered-IPS-Touch-Screen-800x480]
This is the github page for the elecrow 7inch Advanced, I've tried some messing around trying to make my own config but i'm way out of my depth.

The git has lots of pin definitions but I am unsure if they cover everything needed for its own config in OpenHASP

closed 2025-07-18 by fvanroie