HASwitchPlate/openHASP #919: Display ESP32-4848S040 with "Guition ESP32-S3-4848S040" firmware does not work properly: can't connect to WiFi

tankist-git-2, 2025-07-12:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://www.openhasp.com/)
- [x] Update openHASP to the latest version
- [x] Reproduce the issue and describe all steps

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

fvanroie, 2025-07-12:
Check the serial log, that should give you a clue about what's happening and why the connection fails.

fvanroie, 2025-07-14:
The only issue I see in the logs is:
```
[      39.733][122868/133692  8][57964/58448  1] WIFI: association leave
[      39.742][122868/133992  8][57964/58448  1] WIFI: Unknown error (66)
```
That error lead me to the esphome thread mentioning a solution to adjust the wifi power. Several users there report that that fixes their problem. openHASP doesn't really do weird things, it mainly uses the Arduino defaults for ESP32.

1. Good to know, there were some mentions about TP-Link issues and [another user also has a TP-Link](https://github.com/HASwitchPlate/openHASP/issues/791), but that could just be a coincidence... at this point it's not clear yet.
2. I'm trying to find a pattern in order to get to the root cause of this problem. The other user is from the UK, which country are you in or what's the WiFi region configured in your router? There is no setting in openHASP yet for WiFi power or country. We need to find the root cause first before adding UI interfaces etc...
3. You can't be sure until you test it though, at this point it is the only idea I've got. The issue happens with the Guiton 480x480, so it could be some weird combination with hardware & firmware settings. I'm no Wireless expert but I can imagine tweaking the power can yield a better result. It sure worked for the users in the esphome thread I posted, so the issue is not just in openHASP.
4. You can compare your firmware settings with the openHASP wifi settings, maybe there is somehting I overlooked. I'm open to suggestions. Are you able to compile and test the source code locally?

As I'm not able to recreate the specific circumstances that cause this issue, it's hard to anticipate a solution. So far the country and/or power settings are the only lead that might be related to the behavior described.

If you can't compile locally I could push some test settings (guesses) for you to test, but that will be a trial and error.
Let me know what you think.

fvanroie, 2025-07-15:
You can compile the firmware [locally](https://openhasp.com/0.7.0/firmware/compiling/local/) or with [GitPod](https://openhasp.com/0.7.0/firmware/compiling/gitpod/). Home Assistant is not required.

I was thinking of something like `WiFi.setTxPower(WIFI_POWER_8_5dBm)` from Arduino or `esp_wifi_set_max_tx_power()`  or `esp_wifi_set_country_code()` function from the ESP-IDF framework.

Holle88, 2025-07-27:
have the same plate, mine is working fine.

tankist-git-2, 2025-07-27:
Obviously, the issue is how quickly the router issues an IP address. But since this time is not specified by the standards, the firmware must support operation in any case, i.e. this is a bug.

fvanroie, 2025-07-27:
This has been fixed in the commit above. Have you been able to test it? Can you confirm this is resolved?
Re connection is now handled by the Arduino/ESP-IDF framework instead of the every_5_seconds callback.

fvanroie, 2025-07-28:
Could you try also setting `WiFi.persistent(true);` ?