HASwitchPlate/openHASP #927: No background colors for object Label

BaffoBill, 2025-08-05:
Describe the bug

The "bg_color" option has no effect on the object "label"
I am testing Ver. 0.7.0 rc12 and rc13
Is there an explanation for this?

[image]

```json
{"page": 1,"id": 5,"obj": "label","x": 175,"y": 5,"h": 50,"w": 100,"text": "Label","align": 2,"bg_color":"#00FF00","text_color":"#FF0000"}
{"page": 1,"id": 6,"obj": "btn","x": 175,"y": 100,"h": 50,"w": 100,"text": "Button","align": 2,"bg_color":"#00FF00","text_color":"#FF0000"}

fvanroie, 2025-08-06:
This is not a bug. The background of a label is transparent by default. Change the opacity to 255 to make the background color visible.

BaffoBill, 2025-08-08:
Oh, thank you! This is a good news for me!