HASwitchPlate/openHASP #936: Problem with Add Suffix MAC causing OpenHASP integration in Home Assistant to fail

MyJenCz, 2025-08-27:
**Description:**
When connecting an OpenHASP device (ESP32 Sunton display, firmware 0.7.0-rc13, esp32-2432S028) to MQTT, the firmware always appends the MAC suffix to the MQTT client ID (e.g., plate01_2f8fd4).

The Home Assistant OpenHASP integration seems to expect a fixed device ID (like plate01) without the suffix. As a result, the integration setup always fails with "Setup failed, check logs".

**Steps to reproduce:**

Flash firmware 0.7.0-rc13 on an ESP32 display.

Configure MQTT with hostname plate01.

Device connects to broker as plate01_2f8fd4.

Try to add the device in Home Assistant via the OpenHASP integration.

**Integration setup fails.**

Expected behavior:
Integration should either:

accept the suffixed client ID automatically, or

provide an option in firmware to disable Add Suffix MAC.

**Logs:**
MQTT: Connected to broker 192.168.0.226 as clientID plate01_2f8fd4
Integration: Setup failed

**Additional context:**

Tested with firmware 0.7.0-rc12 and 0.7.0-rc13.

Without the HA integration, MQTT publish/subscribe works correctly.

ESPHome has a name_add_mac_suffix option to toggle this behavior; a similar option in OpenHASP firmware would solve the problem.

MyJenCz, 2025-08-27:
Configuration in Home Assistant (configuration.yaml):

openhasp:
  plate01_2f8fd4: {}

or tested as:

openhasp:
  plate01: {}

Both configurations fail with setup errors from the OpenHASP integration:
- "Setup failed, check logs"

OpenHASP MQTT configuration:

Hostname: plate01
Node topic: openhasp/%hostname%/%topic%
Group topic: openhasp/%hostname%/group/%group%/
Broadcast topic: openhasp/%hostname%/broadcast
MQTT Base topic in HA integration suggested: openhasp/plate01
ClientID generated automatically: plate01_2f8fd4 (MAC suffix forced)

Observation:
- The firmware appends the MAC suffix to the MQTT ClientID automatically.
- MQTT discovery messages still use the plain hostname "plate01".
- This mismatch prevents the Home Assistant OpenHASP integration from connecting successfully.

fvanroie, 2025-08-27:
> Problem with Add Suffix MAC causing OpenHASP integration in Home Assistant to fail

How did you reach this conclusion? The MQTT device-id and nodename are two different things. Only the MQTT Broker sees the MQTT device-id. All other MQTT communication is done using the nodename in the topic.

> Integration setup fails.

There must be another cause for this, as the MQTT device-id shouldn't matter at all. Did you check the logs or [enable debugging](https://openhasp.com/0.7.0/integrations/home-assistant/howto/#debugging)?

You need to add an `openhasp` section to the  `configuration.yaml` for the integration to work properly.

>The firmware appends the MAC suffix to the MQTT ClientID automatically.
> MQTT discovery messages still use the plain hostname "plate01".

This is by design, and not a bug.

> This mismatch prevents the Home Assistant OpenHASP integration from connecting successfully.

I really don't think this is the issue, it has worked like this for years and there is no such thing as a *mismatch preventing connection* observed. before.  Please share how you have determined this is the cause of the issue?

fvanroie, 2025-08-29:
No worries, I'm happy you figured it out. Have fun!

closed 2025-08-29 by fvanroie